cnadler86/micropython-camera-API #1: Fails to build on ESP32_GENERIC-SPIRAM due to iram overflow

jonnor, 2024-09-25:
Hi and thank you for this library.

I tried building it on ESP32, in order to run on a LilyGO TTGO T-Camera Mic ESP32. However, the build is failing due to running out of iram. I am using the lastest MicroPython release, v1.23.0 with esp-idf 5.0.4. Build is done with BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM

Default build with mp_camera: `iram0_0_seg' overflowed by 1592 bytes
```
make USER_C_MODULES=../../../../mp_camera/src/micropython.cmake BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM all

/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: micropython.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 1592 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Default build without mp_camera - works. Unsure how many bytes are left in iram...
```
make BOARD=ESP32_GENERIC BOARD_VARIANT=SPIRAM all

micropython.bin binary size 0x17a5f0 bytes. Smallest app partition is 0x1f0000 bytes. 0x75a10 bytes (24%) free.
...
bootloader  @0x001000    23072  (    5600 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1549808  (  481808 remaining)
```

jonnor, 2024-09-25:
I tried to disable Ethernet support to reduce iram usage. A tip from a similar problem with iram overflow at https://github.com/orgs/micropython/discussions/14263#discussioncomment-9047807 

Using sdkconfig.base of
```
CONFIG_ETH_USE_SPI_ETHERNET=n
CONFIG_ETH_SPI_ETHERNET_W5500=n
CONFIG_ETH_SPI_ETHERNET_KSZ8851SNL=n
CONFIG_ETH_SPI_ETHERNET_DM9051=n
```

Still runs out of iram, just by 500 bytes less:
```
sram-cache-issue  -mfix-esp32-psram-cache-strategy=memw && :
/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: micropython.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/jon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 1024 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
I am not sure how to reduce iram usage further in the ESP32 build. Any ideas? Or perhaps there is something this module can do differently to reduce iram usage?

cnadler86, 2024-09-25:
Taanks for trying this out on the esp32 board. I am on holiday for the 10 days, so I will try to find out a solution then.
My first guess would be to disable the ov5640 flag in the header (don't know, if some instructions from this part go into iram).

Perhaps this helps.

jonnor, 2024-09-25:
Thank you for the super quick response. I actually found a solution, documenting below. Enjoy your holiday!

So esp-idf has a whole guide on reducing IRAM usage, at https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/ram-usage.html
Seems that disabling PSRAM cache bug workaround can save 10 kB, which is huge. This is only possible on revision 3 chips - but I think that is almost everything after 2020 or so.  The chip revision will be reported by esptool when flashing. I have `Chip is ESP32-D0WDQ6-V3 (revision v3.0)`.

So changing `ports/esp32/boards/sdkconfig.spiram` to have the following lines, makes the build work. And I can capture from the camera! No idea if the data is sensible yet though - have not checked.

```
CONFIG_SPIRAM_CACHE_WORKAROUND=n
CONFIG_ESP32_REV_MIN_3=y
```

This will probably be useful for others struggling with iram overflow on ESP32.

cnadler86, 2024-09-25:
Great news! Also thank you for the efforts and hints on this problem. I will put this in the documentation then. 

In the examples is a webcam can script that should run. You just need to change your wifi settings and open the esp32 ip.

cnadler86, 2024-10-06:
This issue is also dependent on the IDF-Version. I used 5.2.2 and I could compile without any problems. Don't know exactly, if Micropython (there are some IDF-version logics in the cmake Files) or esp-idf fixed such behavior.
Never the less, I would recommend to update the IDF. 
I will still document thus in the next days.

closed 2024-10-07 by cnadler86

sheetaljain01, 2024-11-14:
> Hi and thank you for this library.
>
> I tried building it on ESP32, in order to run on a LilyGO TTGO T-Camera Mic ESP32. However, the build is failing due to running out of iram. I am using the [...]

> Hi and thank you for this library.
>
> I tried building it on ESP32, in order to run on a LilyGO TTGO T-Camera Mic ESP32. However, the build is failing due to running out of iram. I am using the [...]

Hi, 

Could you share the configuration details for building `ESP32_GENERIC_S3`?  

When I compile using the following command:  

```
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM all
```

I get this error:  

```
CMake Error at CMakeLists.txt:25 (message):  
Invalid MICROPY_BOARD_VARIANT specified: SPIRAM  
```

And when I build using this command:  

```
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3
```

I encounter this error:  

```
#error Camera only works on boards configured with spiram
```

Could you help me resolve this issue?"

jonnor, 2024-11-14:
@sheetaljain01 Hi. I have not tested building for ESP32 S3 yet - only regular ESP32. But the variant for S3 is called `SPIRAM_OCT`. not `SPIRAM`. So that should help with that specific issue, at least.

cnadler86, 2024-11-14:
Hi Jon,
thanks also for answering the question. I also replayed in a new separate issue for this.

PR #4 (2024-10-07): First release candidate
Fix #1.
Stable Version. Minor improvements