cnadler86/micropython-camera-API #5: Should ESP32_GENERIC-SPIRAM work on AI Thinker?

davefes, 2024-10-10:
I get the following:

```
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4892
ho 0 tail 12 room 4
load:0x40078000,len:14896
load:0x40080400,len:4
load:0x40080404,len:3372
entry 0x400805b0
W (62) boot.esp32: PRO CPU has been reset by WDT.
W (62) boot.esp32: WDT reset info: PRO CPU PC=0x400862fb
W (63) boot.esp32: WDT reset info: APP CPU PC=0x40081a81
```

cnadler86, 2024-10-10:
In general it should work, it did by the guy from #2 on a esp32 board. Could you provide more information about this error?
Is it while booting or during some specific code?

It seems to be a watchdog error.

davefes, 2024-10-10:
I had read those issues.
I updated the first post.

The code I am running:
```py
import machine
import time
import gc

gc.collect()

if machine.reset_cause() == machine.SOFT_RESET:
    print('doing a machine.reset()')
    time.sleep_ms(5)
    machine.reset()

print('wait 2 seconds to allow a CTRL-C')
time.sleep(2)

from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling

# Camera construction and initialization
camera = Camera(
    data_pins=[1,2,3,4,5,6,7,8],
    vsync_pin=9,
    href_pin=10,
    sda_pin=11,
    scl_pin=12,
    pclk_pin=13,
    xclk_pin=14,
    xclk_freq=20000000,
    powerdown_pin=-1,
    reset_pin=-1,
    pixel_format=PixelFormat.RGB565,
    frame_size=FrameSize.QVGA,
    jpeg_quality=15,
    fb_count=1,
    grab_mode=GrabMode.WHEN_EMPTY
)

img = camera.capture()
print('picture taken')
```

davefes, 2024-10-10:
Has `camera.init' been left out of the example as I have seen `camera.deinit` in your other examples?

davefes, 2024-10-10:
BTW, I want to run this on a ESP32-CAM

cnadler86, 2024-10-10:
Hi,
You have to specify the right ins from your board. Take a look [here] for the right pins.(https://github.com/adafruit/circuitpython/blob/main/ports/espressif/boards/ai-thinker-esp32-cam/pins.c)

The camera initialization is done during construction. You have the deinit only for the purpose of freeing resources, if you need them. Otherwise you can just delete the camera object and run the garbage collector, and have the same result.
deinit might be not really needed at user level, but i left it there just in case.

davefes, 2024-10-10:
Thank you very much.  I have read comments about doing a `camera.deinit` before WiFi so I have the habit of doing init/deinit/init.

closed 2024-10-10 by davefes

cnadler86, 2024-10-10:
You can just connect to wifi and then construct the camera. The implementation from shariltumin used micropython 1.20, i think, so this might not be needed any more. With the right pin assignment it should work =).

davefes, 2024-10-10:
Originally, I was running cameras on a 2Watt PV panel so only used the WiFi when the camera was triggered (snapshots).  I had a look at the link to CircuitPython and it is beyond my skill level.  I will wait to see how this project develops.