cnadler86/micropython-camera-API #8: ESP32-CAM AI-Thinker green tint

davefes, 2024-10-16:
Trying:
```py
# camera construction and initialization
camera = Camera(pixel_format=PixelFormat.JPEG, frame_size=FrameSize.SXGA, jpeg_quality=85, xclk_freq=20000000, grab_mode=GrabMode.LATEST)

# to try to get rid of the green tint
camera.set_awb_gain(False)
camera.set_agc_gain(0)
camera.set_whitebal(0)

img = camera.capture()
```
doesn't seem to work.

I also tried taking 3 captures, which works for me on another distro.

Taking `whitebal` as an example in the previous distro there were 5 values.  I worked out that there is only `true` and `false` for this implementation.

A table showing the parameters values allowed would be helpful.

cnadler86, 2024-10-17:
Have you already clicked on the link point to the circuitpython documentation in the readme?
There you will find a good description of the possible settings and their possible values. As mentioned also in the readme, you will need to append a "set_" to the respective property. For example: there is whitebal (true/false) and wb_mode (0-4).

I will also try to reproduce a green tint situation (I have it only on bad light conditions) and try some of the tricks from the other Threads and forums.

cnadler86, 2024-10-17:
In my case, I just experience a low green tint in the initial frames, but after ~1s of video stream it disappears.
I used set the wb_mode to 3 or 4 (0 - Auto, 1 - Sunny, 2 - Cloudy, 3 - Office, 4 - Home) and I got rid of my green tint at all.
I can imagine that different targets may deliver different pictures (I have a S3 XIAO board with OV2640), and also camera sensors may have some aging, and of course everyone have a different environment and light around, so it is very difficult to fully reproduce the situation.

davefes, 2024-10-17:
Documentation ... I did focus too much on the source code.  Thank you for the prompt.

The comment regarding `set_` did I not do that properly in my snippet?  I also ran the `get_` on each to verify that the changes had actually happened.

I was not aware that there was `wb_mode` as well as `whitebal`.

`special_effect` I think the numbers are:
 `0:,no effect 1:negative, 2:black and white, 3:reddish,  4:greenish, 5:blue, 6:retro`

Appreciated your comments.

davefes, 2024-10-17:
After quite a bit testing I get acceptable pictures with"
```py
 # camera construction and initialization
camera = Camera(pixel_format=PixelFormat.JPEG, frame_size=FrameSize.SXGA, jpeg_quality=85, xclk_freq=20000000, grab_mode=GrabMode.LATEST)
#camera = Camera(pixel_format=PixelFormat.JPEG, frame_size=FrameSize.SXGA, jpeg_quality=85, grab_mode=GrabMode.WHEN-EMPTY)

camera.set_brightness(-2)

# take 10 frames, rejecting the first nine!!
for x in range(10):
    img = camera.capture()
    time.sleep_ms(50)
```

closed 2024-10-17 by davefes

cnadler86, 2024-10-18:
Happy to hear you solve the problem. Have fun!