cnadler86/micropython-camera-API #9: compiling issues

littlegreenbean33, 2024-10-29:
Hi,

I am using micropython 1.22 and idf 5.1. I am targeting the same device as you XIAO esp32s3

I get the following error compiling.

Frankly no idea how to proceed.

Any help highly appreciated

JC

` 125 |     mp_camera_obj_t *self = mp_obj_malloc_with_finaliser(mp_camera_obj_t, &camera_type);
 88       |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
 89       |                             m_malloc_with_finaliser
 90 /home/julian/tmp/proyectos_python/camera/micropython-camera-API/src/modcamera_api.c:125:58: error: expected expression before 'mp_camera_obj_t'
 91   125 |     mp_camera_obj_t *self = mp_obj_malloc_with_finaliser(mp_camera_obj_t, &camera_type);
 92       |                                                          ^~~~~~~~~~~~~~~
 93 cc1: some warnings being treated as errors
 94 make[3]: *** [esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/build.make:3538: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/julian/tmp/proyectos_python/camera/micropython-camera-API/src/modcamera_api.c.obj] Erro    r 1
 95 make[3]: *** Waiting for unfinished jobs....
 96 make[2]: *** [CMakeFiles/Makefile2:4871: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/all] Error 2
 97 make[1]: *** [Makefile:136: all] Error 2
`

cnadler86, 2024-10-29:
Hi. 

Probably this function was not include in Micropython yet, so there are some possibilities:
- You can change to Micropython >=  1.23
- you can go to the actions tab and use one of the last passing Workflow and download from the artifacts section the Xiao image. Then you only have to flash it and the pins will be mapped. This imag has experimental features but shall work without problems for most of the cases. In this case change to the develop branch for documentation.

In a few weeks I will publish a new release, so you could start with the image from the develop branch and then switch in a few weeks to the stable one, if you experience abnormalities.

littlegreenbean33, 2024-10-29:
I am testing the firmware but I am unable to run the tests or examples.

For example

>>> dir(camera)                                                                                                       │~
['__class__', '__name__', '__dict__', 'aecvalue', 'aelevels', 'agcgain', 'brightness', 'capture', 'capture_bmp', 'capt│~
ure_jpg', 'conf', 'contrast', 'deinit', 'flip', 'framesize', 'init', 'mirror', 'pixformat', 'quality', 'saturation', '│~
speffect', 'whitebalance'] 

Does not include "Camera" with "C" in capitals. 

Parameters for framesize is not working for me either.

Let me know if I can help you on anything documenting or testing. I have interest in having this working.

cnadler86, 2024-10-30:
Hi,

I am not sure what you are sending me, but this are not methods or functions I have in my code (or at least that's not the syntax I am using). Can you please provide the code you are trying to run and the complete output of your IDE? Which camera? And from which workflow run did you got the firmware?

Basically, if you are using the firmware image, you can capture an image by typing:

'''

from camera import Camera, PixelFormat, FrameSize
Cam= Camera(pixel_format=PixelFormat.JPEG)
Cam.capture()
'''

There is indeed an issue with the CameraSettings.py example in Micropython 1.24. Seams to be something with asyncio. I will investigate this and raise an issue in the respective repo. The other examples should work.

littlegreenbean33, 2024-11-02:
I am not skilled to do the compiling myself. Trying your examples I get

>>> cam = Camera(frame_size = FrameSize.VGA,pixel_format=PixelFormat.JPEG)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'data_pins' argument required

 File "<stdin>", line 2, in <module>
NameError: name 'PixeQVGA' isn't defined
pixelformat seem to be
>>> PixelFormat.
GRAYSCALE       JPEG            RGB565          YUV422

I am using the xiao sense with OV2640. Seems like the pinout is mandatory. Can you share this ?

Thanks.

JC

littlegreenbean33, 2024-11-02:
I cannot even run the examples form the firmwares posted. I am using a xiao

 from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling

 cam = Camera(
...     data_pins=[1, 2, 3, 4, 5, 6, 7, 8],
...     vsync_pin=9,
...     href_pin=10,
...     sda_pin=11,
...     scl_pin=12,
...     pclk_pin=13,
...     xclk_pin=14,
... xclk_freq=20000000,  # 20 MHz
...     powerdown_pin=-1,    # Optional
...     reset_pin=-1,        # Optional
...     pixel_format=PixelFormat.RGB565,
...     frame_size=FrameSize.QVGA,
...     jpeg_quality=15,
...     fb_count=1,
...     grab_mode=GrabMode.WHEN_EMPTY
... )
Traceback (most recent call last):
  File "<stdin>", line 16, in <module>
OSError: Camera not found

cnadler86, 2024-11-03:
Hi There.

please read carefully the documentation. This will help you next time. 
I Assume you are using a image from the first release. In this case, the pins need to be adapter to your board. The pins for the XIAO are actually stated in the documentation as example .
Anyway, you can use [this binary image ](https://github.com/cnadler86/micropython-camera-API/actions/runs/11581641705/artifacts/2119825971) (will be available only approx. one day more), flash it and then just use the following:

```
from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling
cam = Camera(frame_size = FrameSize.VGA,pixel_format=PixelFormat.JPEG)
img = cam.capture()
```

For Images from the actions tab, please refer to the [develop branch](https://github.com/cnadler86/micropython-camera-API/tree/develop) for documentation.
If the image is not available any more, you can go to the [actions tab](https://github.com/cnadler86/micropython-camera-API/actions), click on any workflow with a green bubble that is less than 5 days old, search for xiao (Cntl +f), and download the image. 

I am going to close the issue because it should be now clear what to do and the issue is not on software side.

closed 2024-11-03 by cnadler86

littlegreenbean33, 2024-11-03:
Sorry for being so dumb. Let me share the steps I follow

This is my base dir
`
drwxrwx---  9 julian julian 4096 nov  3 09:40 esp-idf 

drwxrwx--- 16 julian julian 4096 nov  3 10:15 micropython

drwxrwx---  7 julian julian 4096 nov  3 12:19 micropython-camera-API
`
I have download and installed latest version of esp-idf
I have cloned micropython v1.24.0
I have clonned esp32_camera

`
~/tmp/proyectos_python/camera$ ls esp-idf/components/esp32-camera/
sdkconfig.flash_4m CMakeLists.txt  driver    idf_component.yml  library.json  README.md  target 
conversions     examples  Kconfig            LICENSE       sensors    test 
`

ESP32_GENERIC_S3_CAMERA is a verbatim copy of ESP32_GENERIC_S3 with the following changes
I have modified mpconfigboard.h, mpconfigboard.cmake , mpconfigvariant_FLASH_4M.cmake

`
~/tmp/proyectos_python/camera/micropython/ports/esp32$ more boards/ESP32_GENERIC_S3_CAMERA/mpconfigboard.h
#ifndef MICROPY_HW_BOARD_NAME
// Can be set by mpconfigboard.cmake.
#define MICROPY_HW_BOARD_NAME               "Generic ESP32S3 module CAM"
#endif
#define MICROPY_HW_MCU_NAME                 "ESP32S3"

// Enable UART REPL for modules that have an external USB-UART and don't use native USB.
#define MICROPY_HW_ENABLE_UART_REPL         (1)

#define MICROPY_HW_I2C0_SCL                 (9)
#define MICROPY_HW_I2C0_SDA                 (8)
#define MICROPY_CAMERA_MODEL_XIAO_ESP32S3       1
`

`
~/tmp/proyectos_python/camera/micropython/ports/esp32$ more boards/ESP32_GENERIC_S3_CAMERA/mpconfigboard.cmake

set(IDF_TARGET esp32s3)

set(SDKCONFIG_DEFAULTS
    boards/sdkconfig.base
    ${SDKCONFIG_IDF_VERSION_SPECIFIC}
    boards/sdkconfig.usb
    boards/sdkconfig.ble
    boards/sdkconfig.spiram_sx
    boards/ESP32_GENERIC_S3_CAMERA/sdkconfig.board
)
`

`
~/tmp/proyectos_python/camera/micropython/ports/esp32$ more boards/ESP32_GENERIC_S3_CAMERA/mpconfigvariant_FLASH_4M.cmake
set(SDKCONFIG_DEFAULTS
    ${SDKCONFIG_DEFAULTS}
    boards/ESP32_GENERIC_S3_CAMERA/sdkconfig.flash_4m
)

`

Then
from ports/esp32 I executed
`
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3_CAMERA clean
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3_CAMERA submodules
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3_CAMERA all
`

Save for the below message all went smooth.
`
[1420/1428] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/home/julian/tmp/proyectos_python/camera/micropython-camera-API/src/modcamera.c.obj
/home/julian/tmp/proyectos_python/camera/micropython-camera-API/src/modcamera.c:294:12: warning: 'map' defined but not used [-Wunused-function]
  294 | static int map(int value, int fromLow, int fromHigh, int toLow, int toHigh) {
      |            ^~~
[1427/1428] Generating binary image from built executable
`

When running the code in a console I get the following results
`
>>> from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling
>>> camera = Camera()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'data_pins' argument required
>>> 
`
Not as expected, as I presumably have xiao default config

Then
`
MicroPython v1.24.0 on 2024-11-03; Generic ESP32S3 module CAM with ESP32S3
Type "help()" for more information.
>>> from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling
>>> 
>>> cam = Camera(
...     data_pins=[1, 2, 3, 4, 5, 6, 7, 8],
...     vsync_pin=9,
...     href_pin=10,
...     sda_pin=11,
...     scl_pin=12,
...     pclk_pin=13,
...     xclk_pin=14,
...     xclk_freq=20000000,  # 20 MHz
...     powerdown_pin=-1,    # Optional
...     reset_pin=-1,        # Optional
...     pixel_format=PixelFormat.RGB565,
...     frame_size=FrameSize.QVGA,
...     jpeg_quality=15,
...     fb_count=1,
...     grab_mode=GrabMode.WHEN_EMPTY
... )
Traceback (most recent call last):
  File "<stdin>", line 16, in <module>
OSError: Camera not found
`

Again not what I expected.

Can you help me out to figureout what am I missing ?

Thanks

littlegreenbean33, 2024-11-03:
Adding 

`#define MICROPY_CAMERA_PIN_D0       (15)
#define MICROPY_CAMERA_PIN_D1       (17)
#define MICROPY_CAMERA_PIN_D2       (18)
#define MICROPY_CAMERA_PIN_D3       (16)
#define MICROPY_CAMERA_PIN_D4       (14)
#define MICROPY_CAMERA_PIN_D5       (12)
#define MICROPY_CAMERA_PIN_D6       (11)
#define MICROPY_CAMERA_PIN_D7       (48)
#define MICROPY_CAMERA_PIN_PCLK     (13)
#define MICROPY_CAMERA_PIN_VSYNC    (38)
#define MICROPY_CAMERA_PIN_HREF     (47)
#define MICROPY_CAMERA_PIN_XCLK     (10)
#define MICROPY_CAMERA_PIN_PWDN     (-1)
#define MICROPY_CAMERA_PIN_RESET    (-1)
#define MICROPY_CAMERA_PIN_SIOD     (40)  // SDA
#define MICROPY_CAMERA_PIN_SIOC     (39)  // SCL
#define MICROPY_CAMERA_XCLK_FREQ    (20000000)  // Frequencies are normally either 10 MHz or 20 MHz
#define MICROPY_CAMERA_FB_COUNT     (2)   // Usually the value is between 1 (slow) and 2 (fast, but more load on CPU)
#define MICROPY_CAMERA_JPEG_QUALITY (10)  // Quality of JPEG output. 0-63 lower means higher quality. Definition will change in the future
#define MICROPY_CAMERA_GRAB_MODE    (1)   // 0=WHEN_EMPTY (might have old data, but less resources), 1=LATEST (best, but more resources)`

results in 
`
MicroPython v1.24.0 on 2024-11-03; Generic ESP32S3 module CAM with ESP32S3
Type "help()" for more information.
>>> from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling
>>> camera = Camera()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unknown error 0xffffffff
`

If I manually define the pins, I get the same error as before with Camera not defined.

cnadler86, 2024-11-03:
Have you changed your camera branch to develop?

And why are you not simply using the firmware image that I shared over the link before? This will be the easiest way.

And one more note: the Xiao has octal psram, so you will need to pass it as a board_variant

cnadler86, 2024-11-03:
> Adding  in mpconfigboard.h
> 
> 
> 
> `
> 
> #define MICROPY_CAMERA_PIN_D0       (15)
> 
> #define MICROPY_CAMERA_PIN_D1       (17)
> 
> #define MICROPY_CAMERA_PIN_D2       (18)
> 
> #define MICROPY_CAMERA_PIN_D3       (16)
> 
> #define MICROPY_CAMERA_PIN_D4       (14)
> 
> #define MICROPY_CAMERA_PIN_D5       (12)
> 
> #define MICROPY_CAMERA_PIN_D6       (11)
> 
> #define MICROPY_CAMERA_PIN_D7       (48)
> 
> #define MICROPY_CAMERA_PIN_PCLK     (13)
> 
> #define MICROPY_CAMERA_PIN_VSYNC    (38)
> 
> #define MICROPY_CAMERA_PIN_HREF     (47)
> 
> #define MICROPY_CAMERA_PIN_XCLK     (10)
> 
> #define MICROPY_CAMERA_PIN_PWDN     (-1)
> 
> #define MICROPY_CAMERA_PIN_RESET    (-1)
> 
> #define MICROPY_CAMERA_PIN_SIOD     (40)  // SDA
> 
> #define MICROPY_CAMERA_PIN_SIOC     (39)  // SCL
> 
> #define MICROPY_CAMERA_XCLK_FREQ    (20000000)  // Frequencies are normally either 10 MHz or 20 MHz
> 
> #define MICROPY_CAMERA_FB_COUNT     (2)   // Usually the value is between 1 (slow) and 2 (fast, but more load on CPU)
> 
> #define MICROPY_CAMERA_JPEG_QUALITY (10)  // Quality of JPEG output. 0-63 lower means higher quality. Definition will change in the future
> 
> #define MICROPY_CAMERA_GRAB_MODE    (1)   // 0=WHEN_EMPTY (might have old data, but less resources), 1=LATEST (best, but more resources)
> 
> `
> 
> 
> 
> results in 
> 
> 
> 
> `
> 
> MicroPython v1.24.0 on 2024-11-03; Generic ESP32S3 module CAM with ESP32S3
> 
> Type "help()" for more information.
> 
> >>> from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling
> 
> >>> camera = Camera()
> 
> Traceback (most recent call last):
> 
>   File "<stdin>", line 1, in <module>
> 
> RuntimeError: Unknown error 0xffffffff
> 
> `
> 
> 
> 
> If I manually define the pins, I get the same error as before with Camera not defined.
> 
> 

This would be the right way. Just see the octal psram sdkconfig and add the lines from there to your sdkconfig. Then you should also have bull psram

littlegreenbean33, 2024-11-03:
> Have you changed your camera branch to develop?
> 
> And why are you not simply using the firmware image that I shared over the link before? This will be the easiest way.
> 
> And one more note: the Xiao has octal psram, so you will need to pass it as a board_variant

No, I have not changed. I probably have an esp32 eye cam. Will try.

I tried the pre-build firmwares. WIll try again to see if I am missing something. Tried both S3 versions.

I know the variant thing, but I was trying to first get it running

littlegreenbean33, 2024-11-03:
[image]

same results

littlegreenbean33, 2024-11-03:
I just tested with the firmware you mention on you comments above (https://github.com/cnadler86/micropython-camera-API/actions/runs/11581641705/artifacts/2119825971)

`from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling

# Camera construction and initialization
# These pins are just examples and if you use them just like that will get a watchdog error. Adapt them to your board!
camera = Camera(
    data_pins=[1,2,3,4,5,6,7,8],
    vsync_pin=9,
    href_pin=10,
    sda_pin=11,
    scl_pin=12,
    pclk_pin=13,
    xclk_pin=14,
    xclk_freq=20000000,
    powerdown_pin=-1,
    reset_pin=-1,
    pixel_format=PixelFormat.RGB565,
    frame_size=FrameSize.QVGA,
    jpeg_quality=15,
    fb_count=1,
    grab_mode=GrabMode.WHEN_EMPTY
)`

produces

`>>> from camera import Camera, GrabMode, PixelFormat, FrameSize, GainCeiling
>>>
>>> # Camera construction and initialization
>>> # These pins are just examples and if you use them just like that will get a watchdog error. Adapt them to your board!
>>> camera = Camera(
...     data_pins=[1,2,3,4,5,6,7,8],
...     vsync_pin=9,
...     href_pin=10,
...     sda_pin=11,
...     scl_pin=12,
...     pclk_pin=13,
...     xclk_pin=14,
...     xclk_freq=20000000,
...     powerdown_pin=-1,
...     reset_pin=-1,
...     pixel_format=PixelFormat.RGB565,
...     frame_size=FrameSize.QVGA,
...     jpeg_quality=15,
...     fb_count=1,
...     grab_mode=GrabMode.WHEN_EMPTY
... )
Traceback (most recent call last):
  File "<stdin>", line 16, in <module>
OSError: Camera not found
`

However,

This time I can dor
`>>> import camera
>>> cam = camera.Camera()
>>> cam.init()
>>> c = cam.capture()
>>> c
<memoryview>
>>>
`

so it is working. I tried to dump c into a jpg file, but it is not readable, not the right format.

I'll keep playing to what can I learn.
I'll report to you here if you do not mind.

cnadler86, 2024-11-03:
Happy to hear that. You can also take a look at the examples (you need to provide your ssid and wifi password). Here I want to note that the camera settings example may crash since there may be an issue with the new asyncio library in Micropython 1.24. But the simple webcam example is working. 

And on how to dump a memoryview into a jp**e**g file or other things, please ask chatgpt. You can also promt the example to it to explain you what the code does. Please read the documentation and ask chatgpt before raising issues.

littlegreenbean33, 2024-11-03:
eventually all works when I compile this way

make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3_CAMERA BOARD_VARIANT=SPIRAM_OCT all

see the adding BOARD_VARIANT=SPIRAM_OCT

I leave it here as lessons learned for any body starting form scratch.....