cnadler86/micropython-camera-API #13: Micropython build problem ESP32_GENERIC_S3

sheetaljain01, 2024-11-14:
"Hi, 

Thanks for this library

Could you share the configuration details for building `ESP32_GENERIC_S3`, I am trying to build custom firmware?  

When I compile using the following command:  

```
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM all
```

I get this error:  

```
CMake Error at CMakeLists.txt:25 (message):  
Invalid MICROPY_BOARD_VARIANT specified: SPIRAM  
```

And when I build using this command:  

```
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3
```

I encounter this error:  

```
#error Camera only works on boards configured with spiram
```

Could you help me resolve this issue?"  

---

Let me know if you need further assistance!

cnadler86, 2024-11-14:
Here are the variants
[image]

There is no special configuration for neither of the boards.

This depends on the type of spiram your boards has access to. Either sx (that's the 3s standard) or octal (this is a board vatiant). 

I will suggest to first try the precompiled FW for generic S3 and generic s3-octal and see which of both has the full ram access with the garbage collector. Then you know if you hace a octal one or not. 

You probably need to perform a "clean" before recompiling.  So basically call the clean command, the submodules and the build, everything only with the board=S3, then the compilation should work. 

The pipeline is passing and all biards supported by the library manage to produce a binary image, so the issue is not in the library.

cnadler86, 2024-11-14:
So basically call this here:

```
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3 clean
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3 submodules
make USER_C_MODULES=../../../../micropython-camera-API/src/micropython.cmake BOARD=ESP32_GENERIC_S3 all
```

or add `BOARD_VARIANT=SPIRAM_OCT` to all commands in case of OCTAL ram type

closed 2024-11-16 by cnadler86

sheetaljain01, 2024-11-18:
Thanks for your help!  

I tested your release on my hardware using [ESP32_GENERIC_S3-SPIRAM_OCT.zip](https://github.com/cnadler86/micropython-camera-API/releases/download/v0.2.0/firmware-ESP32_GENERIC_S3-SPIRAM_OCT.zip), and it worked perfectly.  

However, when I create my custom firmware, the firmware build process is successful, but I encounter an error when the camera starts streaming after obtaining the IP address.  

Could you help me resolve this issue?"

[image]

cnadler86, 2024-11-18:
I think the problem is on your build, probably on your commands. Are you building the submodules with the right board and Variant as I pointed out previously?

You cat take a look at the GitHub Workflows in the repo for additional information and ask chatgpt.

Otherwise I don't have a clue, sorry.

sheetaljain01, 2024-11-18:
Can you share the IDF version that was used in the recent release?

cnadler86, 2024-11-18:
5.2.3