cnadler86/micropython-camera-API #18: Documentation for format conversions

jonnor, 2024-12-10:
Hi @cnadler86 - thanks again for the great work on this library. The range of prebuilt binaries are amazing!

I was looking at the options for having the camera operate in JPEG, but then get the image pixel data out. As it seems that this is the recommended approach over sending raw data from camera, especially at higher resolutions?

esp32 camera has functions for doing the conversions, such as `fmt2rgb888`. And actually it seems that these are used/available in this library? Looking at the code, it seems that the convert() function can take an optional out_format argument? But I have not seen this mentioned in any of the documentation?

It would be great to have this briefly covered in the documentation, and maybe a simple example?

cnadler86, 2024-12-11:
Hi Jon,
Great Hearing from you once again my friend. Thanks for the feedback and indeed I forgot to document this. I adapted the documentation a little bit (see capture section). I will provide some more insight on that and the library needs also some improvement in this functionality to get properly GC. Will do it in the next days. 

As far as I tested it, the conversation from jpeg works faster on lower frame sizes. On bigger once it is slower, probably because of psram speed. There is a benchmark example for image conversion. 

This might not be the right place to comment on, but I am looking forward on testing the emlearn with the camera, probably on christmas vacations. 

Let me know if this short piece of documentation helps you. 

Regards.

jonnor, 2024-12-14:
Thanks for the quick response. I guess together with the code, it is probably enough to get started with.

Actually, when planning to use this - I realize I have a request for a change in this functionality. Can you change (or add to) the API, such that there is a separate function call for the format conversion? That way one can access both the JPEG, and the converted image frame. My use-case for that is to be able to save the JPEG as data for ML training/validation/testing, while running the neural network / ML model on the converted image data. Without such a split, I would have to either decode the JPEG myself or to re-encode as JPEG after the conversion. Both are wasteful, and there are no good/efficient JPEG encoder/decoders for MicroPython at the moment, so would be lost of additional work. 

I think such an API could be simpler to implement wrt GC as well. The function could take an input, as well as an array (of correct size) to put the output in. That way it is zero-allocation inside the module, and one can avoid the extra allocation on each frame by passing in a pre-created array.

Pseudo-code for example usage
```python
converted = array.array('B', range(width*height*3))

raw = camera.capture()
camera.convert(raw, converted, format=FOO)
# converted now contains the converted data (assuming no exception)
```

cnadler86, 2024-12-15:
Hi,

I was just thinking, if there is a use case fox exact that situation. Back then, when I started integrating the image conversion I thought there is no use case for having both and then implemented the conversion directly in the capture function. But it seems there is one =).

I will add this function the next time I find time to do this. The idea of having the memory allocation at micropython level is nice. I will do it like that and add a short example also.

Regards,

closed 2024-12-17 by cnadler86

cnadler86, 2024-12-17:
Hi there,
at the end I took another approach for the functionality because one image conversion function in the esp32-camera driver allocates memory, so to be consistent I am doing the allocation for all conversions. Now, it shall be properly GC also. Reganrs

jonnor, 2024-12-18:
@cnadler86 the split convert() functionality looks super. Thank you very much! Hope to play with this in the holidays

cnadler86, 2025-02-24:
Hi @jonnor,

I had time to implement a wrapper for the new esp jpeg decoder/encoder. Here the [link](https://github.com/cnadler86/mp_jpeg).
I am considering throwing the conversion functionality out of the camera api. With the external decoder should be enough, I think (don't see any use cases in converting RGB565 to RGB888, if you have a fast JPEG decoder/encoder).

I performed some benchmarks and this jpeg decoder is more or less 1.8x faster than the actual one. On small images, you will not notice difference at the FPS, because the camera is the limitation, but it uses less resources, so you will have more for the ML-model.

commit e462cac2 (2024-12-17): Img conv fun (#21)

Adds more functionality and improves the performance in image conversion.
Fix #18