cnadler86/micropython-camera-API #19: Example?

woofwoof, 2024-12-12:
Allo,

Humble apologies. Maybe I just need to delve right in but I was wondering if it's possible to provide at least one worked through example please? I would like to try your code with ESP32S3 Sense. Could you please tell me if that's possible before I try...Maybe then I can help illustrate the whole approach?

With thanks.

cnadler86, 2024-12-12:
See the examples folder. And yes, it should work, but please read the documentation. If you want more insides in the example code ask chatgpt

woofwoof, 2024-12-12:
Allo,

Thank you for such a quick reply! It's good to know it should work with ESP32S3 Sense. I took a look at all of your documentation. I don't use chatgpt. I'll schedule some time at the weekend and if if get chance document a worked through example.

All best.

woofwoof, 2024-12-18:
Hello,

Thanks again for your efforts. I got the Xiao ESP32S3 Sense camera working with the following initialisaion. There's a little lag presently on the webcam example but that doesn't worry me as I'll most likely be using still images eventually.

```
cam = Camera(
    data_pins=[15,17,18,16,14,12,11,48],
    vsync_pin=38,
    href_pin=47,
    sda_pin=40,
    scl_pin=39,
    pclk_pin=13,
    xclk_pin=10,
    xclk_freq=20000000,
    powerdown_pin=-1,
    reset_pin=-1,
    pixel_format=PixelFormat.JPEG,
    frame_size=FrameSize.VGA,
    init=False
)
```
All the best. WW.

cnadler86, 2024-12-18:
Thanks for the feedback. 
But, why are you not using the precompiled firmware for the Xiao? You can then skip passing the whole pin assignments.

woofwoof, 2024-12-18:
Hello,

No problem. I did use the xiao firmware but without assigning the pins the camera wasn't displaying an image in the example web cam application. Is the firmware for the Sense camera I wonder?

Thanks.

cnadler86, 2024-12-18:
Yes, it should. I worked for me without a problem (I have the same)

woofwoof, 2024-12-18:
Hmm... sorry then I don't know what's different. Although I did read that there are two versions of the sense board. Who knows. It definitely works with the pins explicitly assigned though. All the best.

cnadler86, 2024-12-18:
Are exactly the same. So, probably the issue is somewhere else. But they r it works it's fine. Please close the issue then.

closed 2024-12-18 by woofwoof