cnadler86/micropython-camera-API #20: Better way of handling properties

kdschlosser, 2024-12-16:
instead of adding a bunch of functions to deal with properties there is a better way to go about doing this...

Here is an example..

```c
static mp_obj_t mp_camera_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *all_args)
{
    // do new code here
}

// this will check any parent classes for methods and attributes as well
static void call_methods(mp_obj_t obj, qstr attr, mp_obj_t *dest)
{
    const mp_obj_type_t *type = mp_obj_get_type(obj);
    while (MP_OBJ_TYPE_HAS_SLOT(type, locals_dict)) {
        assert(MP_OBJ_TYPE_GET_SLOT(type, locals_dict)->base.type == &mp_type_dict);
        mp_map_t *locals_map = &MP_OBJ_TYPE_GET_SLOT(type, locals_dict)->map;
        mp_map_elem_t *elem = mp_map_lookup(locals_map, MP_OBJ_NEW_QSTR(attr), MP_MAP_LOOKUP);
        if (elem != NULL) {
            mp_convert_member_lookup(obj, type, elem->value, dest);
            break;
        }
        if (MP_OBJ_TYPE_GET_SLOT_OR_NULL(type, parent) == NULL) {
            break;
        }
        // search parents
        type = MP_OBJ_TYPE_GET_SLOT(type, parent);
    }
}

static void camera_attr(mp_obj_t self_in, qstr attr, mp_obj_t *dest)
{
    mp_camera_obj_t *self = MP_OBJ_TO_PTR(self_in);

    if (dest[0] == MP_OBJ_NULL) {
        // get attribute
        switch(attr)
        {
            case MP_QSTR_attr_name1:
                // code to collect attribute from  mp_camera_obj_t structure
                dest[0] = // set attribute value to be returned. use one of the mp_obj_new_* functions to convert a ctype to a python type 
                break;
            case MP_QSTR_attr_name2: 
                // code to collect attribute from  mp_camera_obj_t structure
                dest[0] = // set attribute value to be returned. use one of the mp_obj_new_* functions to convert a ctype to a python type 
                break;
            default: 
                call_methods(self_in, attr, dest);
                
            // if dest[0] does not get populated at all then AttributeError will be raised
        }
    } else {
        if (dest[1])
        {
            // set attribute
            switch(attr)
            {
                case MP_QSTR_attr_name1: 
                    // dest[1] contains the mp_obj_t structure that holds the value.
                    // there are many functions in micropython to convert the python data into c data
                    break;
                case MP_QSTR_attr_name2:
                    // conversion from python type to c type here
                    break;
                
                default: return;
            }
            // set dest[0] to MP_OBJ_NULL to not raise an AttributeError exception
            dest[0] = MP_OBJ_NULL;
        }
    }
}

// few methods
static mp_obj_t camera_capture(size_t n_args, const mp_obj_t *args)
{
    // do method code here
}

static MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(camera_capture_obj, 1, 2, camera_capture);

static mp_obj_t camera_reconfigure(size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args)
{
    // do method code here
}

static MP_DEFINE_CONST_FUN_OBJ_KW(camera_reconfigure_obj, 1, camera_reconfigure);

static const mp_rom_map_elem_t camera_camera_locals_table[] = {
    { MP_ROM_QSTR(MP_QSTR_reconfigure), MP_ROM_PTR(&camera_reconfigure_obj) },
    { MP_ROM_QSTR(MP_QSTR_capture), MP_ROM_PTR(&camera_capture_obj) },
};

static MP_DEFINE_CONST_DICT(camera_camera_locals_dict, camera_camera_locals_table);

static MP_DEFINE_CONST_OBJ_TYPE(
    camera_type,
    MP_QSTR_Camera,
    MP_TYPE_FLAG_NONE,
    make_new, mp_camera_make_new,
    attr, camera_attr,
    locals_dict, &camera_camera_locals_dict,
);
```

kdschlosser, 2024-12-17:
auto complete in thonny is going to be dependent on a stub file being available.

cnadler86, 2024-12-17:
Will try this out next time my kids wake me up at 4am 🤪

cnadler86, 2025-01-04:
I thought about your suggestion, but changing this won't would mean to change the architecture a lot. So at the moment I will leave it as it is and take this into consideration for further developments. Thanks anyway.

closed 2025-10-26 by kdschlosser