cnadler86/micropython-camera-API #24: Documentation fixup for build commands

anicolao, 2025-02-22:
In the `README.md` it might be nice to note that the path to `micropython.cmake` **deliberately** includes an extra `../` level because of where the `CMakeLists.txt` file is for the esp32 port. That way people who aren't just blindly copying and pasting these commands are less likely to encounter the error that the camera module is not found.

cnadler86, 2025-02-22:
Hi,

Thanks for the feedback.

It depends where you locate the camera-api-folder in Relation to the Micropython path.  If you locate them in parallel (at the same level), the commands should work (at least they worked the last time I build it with Micropython 1.24.1.). I can add this information to make it more clear. Still, it is just a suggestion and the build is in control of the one who is performing it.

closed 2025-02-24 by cnadler86

anicolao, 2025-02-24:
I think you misunderstood my suggestion. The commands you have shown work, but they point at a spot that doesn't exist. This is in accordance with the documentation for building the esp32 port, which reads:

> ```
> cd my_project/micropython/ports/esp32
> make USER_C_MODULES=../../../../modules/micropython.cmake
> ```
> Note that the esp32 port needs the extra .. for relative paths due to the location of its main CMakeLists.txt file. You can also specify absolute paths to USER_C_MODULES.

So your listed directions are correct, but I think you should include the note about the path looking wrong **on purpose** since someone who is paying attention may assume you've made an error and then have to dig to discover why the path looks incorrect.