cnadler86/micropython-camera-API #25: driver doesn't run with 5.4

anicolao, 2025-02-23:
When compiled against esp-idf v5.4 the firmware crash loops. Is this a known issue or am I building it wrong?

cnadler86, 2025-02-23:
Micropython does not support 5.4 yet. As fast as I know 5.3 should be supported, but Micropython 1.24.1 still uses 5.2.2. The issue is probably in Micropython.

anicolao, 2025-02-24:
I do not believe we are reading the same documentation. A `git blame` of `ports/esp32/README.md` shows:

> ```
> e017f276f7 (Damien George 2021-02-14 17:51:31 +1100  30) The ESP-IDF changes quickly and MicroPython only supports certain versions.
> 865a4c8bf6 (IhorNehrutsa  2025-01-21 08:48:25 +0200  31) Currently MicroPython supports v5.2, v5.2.2, v5.3 and v5.4.
> ```

I have built and run micropython without your package successfully for esp32s3 and wroom.

cnadler86, 2025-02-24:
Yes, thanks for clarifying. I missed that PR!! 🙃

I will check it out next time I work on some updates of the API. IDF 5.4 shall also bring a new scab driver for the sensor, so it might also be more stable.

anicolao, 2025-02-24:
I didn't dig deep but it seems there is some change in how the ADC pins are accessed that is checked at runtime and if the old way and the new way are both used it causes problems. Please bear in mind that I have almost no experience with esp-idf and micropython so I may have been well off on the wrong track.

cnadler86, 2025-02-24:
I don't think this has to do with the ADCs. There  is a [issue on the esp32 camera driver](https://github.com/espressif/esp32-camera/issues/727) that might be related to this one.

Try changing the line 43 of [sdkconfig.base](https://github.com/micropython/micropython/blob/master/ports/esp32/boards/sdkconfig.base) to:
CONFIG_PM_ENABLE=n
(Don't know about side effects, probably the energy consumption will increase)

cnadler86, 2025-02-24:
And also one question: what is the version of the esp32 camera driver you are using? Or how did you integrate it in your build, by checking it out or over a the IDF-component file?

cnadler86, 2025-03-01:
So, found the problem. Micropython uses the old i2c driver and the esp32-camera driver uses the new one when using idf >=5.4. The problem is, both drivers cannot coexist.

anicolao, 2025-03-02:
I must have mismatched the message about incompatible drivers with threads I found about a similar error message for ADC pins when it was really about i2c.

cnadler86, 2025-03-02:
I don't know how much was introduced in idf 5.4, but Micropython may take a while to migrate all new drivers. So, there are 2 possibilities:
- use idf 5.3
- use the old sccb driver in the esp32 camera library (I tried this and it works fine under 5.4).

commit 7fbb4143 (2025-03-21): asyncio support for capture method and get rid of convert methods (#28)

Adds `acamera `instance supporting async capture method 
Use [mp_jpeg](https://github.com/cnadler86/mp_jpeg) instead of convert
methods.

Fix #25
Fix #26
PR #28 (2025-03-21): asyncio support for capture method and get rid of convert methods
Adds `acamera `instance supporting async capture method 
Use [mp_jpeg](https://github.com/cnadler86/mp_jpeg) instead of convert methods.

Fix #25
Fix #26
closed 2025-03-21 by cnadler86