cnadler86/micropython-camera-API #26: Add new jpeg decoder and get rid of current image converter

cnadler86, 2025-02-24:
The new jpeg decoder delivers aprox. 2x the performance of the actual jpeg decoder.

anicolao, 2025-02-24:
It's probably no help to you at all but in my view the `cam.convert` API shouldn't exist in this package at all. There should be some different module for doing image conversions that the user can use with any camera or image processing application. Maybe a stripped down `pil` or `upil` option.

closed 2025-03-21 by cnadler86

cnadler86, 2025-05-15:
https://github.com/cnadler86/mp_jpeg

I thought you already try this out. It is included in the last release binaries. If you want to build yourself just follow the instructions.

PsuFan, 2025-05-15:
Haha yeah I did, didn’t remember I was talking to you specifically. It works great, I need them combined now, stands to reason others might too. I found the ESP32-S3 CAM EYE. I’m sure I can figure out how to compile them together eventually, just wanted to make sure it didn’t already exist.

commit 7fbb4143 (2025-03-21): asyncio support for capture method and get rid of convert methods (#28)

Adds `acamera `instance supporting async capture method 
Use [mp_jpeg](https://github.com/cnadler86/mp_jpeg) instead of convert
methods.

Fix #25
Fix #26
PR #28 (2025-03-21): asyncio support for capture method and get rid of convert methods
Adds `acamera `instance supporting async capture method 
Use [mp_jpeg](https://github.com/cnadler86/mp_jpeg) instead of convert methods.

Fix #25
Fix #26