cnadler86/micropython-camera-API #31: capture returns none while cam is initialised

FredrikMorstad, 2025-05-02:
**Describe the bug**
Camera becomes None or cannot init properly after multiple attempt to capture image

I have this endpoint using microdot
```python
@app.route('/picture')
async def picture(request):
    # try:
    start_time = time.time()
    cam.init()
    # print(dir(cam))
    buff = None
    if cam is not None:
        while buff is None:
            buff = cam.capture()
            print(buff)
            cam.free_buffer()
    else:
        return 500

    cam.deinit()
    if buff is None:
        return 500

    end_time = time.time()

    elapsed = end_time-start_time
    print(f"capture time: {elapsed:.3f}")
    return buff, 200, {'Content-Type': 'image/jpeg', 'Content-Length': len(buff)}
```

and every time I run the server the capture method randomly returns None from the capture method. I used 
this [driver](https://github.com/lemariva/micropython-camera-driver) and had no problems where the capture method could not capture an image. Generally the capture method returns an image but randomly it returns none. After retrying for about 8 seconds it returns an image but the next request always fails with this exception
```
Traceback (most recent call last):
  File "microdot/microdot.py", line 1394, in dispatch_request
  File "microdot/microdot.py", line 45, in invoke_handler
  File "<stdin>", line 50, in picture
OSError: (-261, 'ESP_ERR_NOT_FOUND')
```

**To Reproduce**
Code to reproduce
```python
from microdot import Microdot
from camera import Camera, PixelFormat, Version

app = Microdot()

cam = Camera(
    pixel_format=PixelFormat.JPEG,
    data_pins=[5,18,19,21,36,39,34,35],
    vsync_pin=25,
    href_pin=23,
    sda_pin=26,
    scl_pin=27,
    pclk_pin=22,
    xclk_pin=0,
    xclk_freq=20000000,
    powerdown_pin=32,
    reset_pin=-1,
)

@app.route('/picture')
async def picture(request):
    cam.init()
    buff = None
    if cam is not None:
        while buff is None:
            buff = cam.capture()
            print(buff)
            cam.free_buffer()
    else:
        return 500

    cam.deinit()
    if buff is None:
        return 500
    return buff, 200, {'Content-Type': 'image/jpeg', 'Content-Length': len(buff)}

app.run(debug=True)
```

**Expected behavior**
A clear and concise description of what you expected to happen.
capture method should be consistent when camera is initialised

**Environment:**
 - Board: e.g. ESP32-CAM 
 - Camera drivers version: 28035e0

**Additional context**
Add any other context about the problem here.

cnadler86, 2025-05-02:
first of all, please read the documentation and take a look at the examples. 

You don't need to init the camera (only if you pass the init flag = false at construction)
Second, you don't need to free the buffer.

cnadler86, 2025-05-02:
If you want to stay with your impelemation, pass init flag with false and convert you buffer to bytes before freeing the Buffer

FredrikMorstad, 2025-05-02:
I have updated the code and it still has the same issues.  I originally initialised the camera at startup and then used a global cam variable but this resulted in the same problem. So I tried an approach more similar to the example but this also results in the same error. Also the can get 5 or 6 captures resulting in none which is very strange

FredrikMorstad, 2025-05-02:
Update:
I moved back to having a global camera instance and it seems to be working now. The only problem is that sometimes the capture method can sometimes use 1-10 tries to get an image from the capture method. But that is probably not the responsibility of this lib to find out why

closed 2025-05-02 by FredrikMorstad