cnadler86/micropython-camera-API #33: Micropython Error

Rainning-Tacos1, 2025-05-18:
**Error:**
```
MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
OSError: (-261, 'ESP_ERR_NOT_FOUND')
>>> 
```
**Code:**
```py
import network
import socket
import time

from camera import Camera, FrameSize, PixelFormat
# Cam Config

cam = Camera(frame_size = FrameSize.VGA,pixel_format=PixelFormat.JPEG,init=False,
    data_pins=[1,2,3,4,5,6,7,8],
    vsync_pin=9,
    href_pin=10,
    sda_pin=11,
    scl_pin=12,
    pclk_pin=13,
    xclk_pin=14,
    xclk_freq=20000000,
    powerdown_pin=-1,
    reset_pin=-1             
)
```

**Setup:**
I am using wsl.
I am using esp-idf v5.4.1
I am using the `micropython-camera-API` and `mp_jpeg`.

**Steps:**
After cloning micropython and the libraries I followed allong the `README.md`:

I did not find a `idf_component.yml` in `micropython/ports/esp32/main_esp32s3` but rather in `micropython/ports/esp32/main`

I included the line `#define MICROPY_CAMERA_MODEL_XIAO_ESP32S3 1` as well as the pin configuration provided for the board.

I then changed some configurations to allow PSRAM

and then built with:
```
idf.py set-target esp32s3
idf.py build
```
It built successfully, and in micropython I am able to `import camera` and `import jpeg`
but when I try one of the examples(`SimpleWebCam.py`)
the code complains about the data pins beeing required
`TypeError: 'data_pins' argument required`
After adding the pins and re-running the code I get the error described at the beginning of this issue.

If you need more info let me know.

cnadler86, 2025-05-19:
Why are you not using the precompiled binary? It has already pin definitions, mp_jpeg and PSRAM.

https://github.com/cnadler86/micropython-camera-API/releases/download/v0.4.0/mpy_cam-v1.25.0-XIAO_ESP32S3.zip

Rainning-Tacos1, 2025-05-19:
I appreciate the binary but I'd rather compile it by hand.

cnadler86, 2025-05-20:
The pins you are using are just an example (as started in the documentation, BTW). 

You probably didn't define the camera board macro in the right file. You have to use the right mpconfigboard.h file (in your case the one of the s3).

Rainning-Tacos1, 2025-05-22:
Sorry for the late response; I changed the pin definitions on the python file to the correct pins and I get no errors.

Unfortunately, I need to specify the pins even though I declared them on the `mpconfigboard.h` when compiling micropython

cnadler86, 2025-05-23:
Actually, you only need the line `#define MICROPY_CAMERA_MODEL_XIAO_ESP32S3 1` and the subsequent pins are not needed any more.  Having the pins defined in here should not bother.

I think you are building the library not correct (follow the documentation), because I have a xioa and everything works fine, so the issue is not in the library.

closed 2025-05-23 by cnadler86