cnadler86/micropython-camera-API #38: Rare Init Issue After Many Reboots

PsuFan, 2025-07-15:
**Describe the bug**
OSError: Failed to capture initial frame. Construct a new object with appropriate configuration.

**To Reproduce**
cam = Camera(pixel_format=PixelFormat.JPEG, frame_size=FrameSize.QVGA, jpeg_quality=90, fb_count=2, grab_mode=GrabMode.LATEST)

**Expected behavior**
Not crash

**Environment:**
 - Board: ESP32-S3 CAM EYE (https://www.amazon.com/dp/B0F4DFFWZV)
 - Camera drivers version: 1.25.0 CAM EYE Firmware Release

**Additional context**
As promised, works great for a long while then overheats or runs out of memory or something. Usually doesn't start working even if I unplug it and plug it back it.

cnadler86, 2025-07-15:
Thanks. Can you please post the full code?

PsuFan, 2025-07-16:
Not really, it’s 800 lines of unrelated stuff. What are you going to see in there?

cnadler86, 2025-07-16:
The problem is, workout the coffee or is impossible to diagnose the error. It could be that you are using pins of the camera for other stuff. Running out of memory, problems with the wifi, Interrupts, etc.

cnadler86, 2025-07-16:
puh. This can be anything. But if you create the camera object at the beginning. Then the problems might decrease. Also if you trigger manually garbage collection after jpeg conversion this might increase the free memory.
Power supply or overheating might also be a thing, because you are using the camera driver at its highest stress.

cnadler86, 2025-07-17:
then something is interfering with the camera driver or you are running Out of memory. Print the amount of memory 9n every loop and wait until it fails. Also, are you using hw timers?

cnadler86, 2025-07-23:
I was just doing some other things and i had an issue similar to that what you are having. I solved it with a cam.deinit() in a finally statement. The point is, that a new camera object doesn't know that another object already initialized the camera peripherals. Might this be your problem?

PsuFan, 2025-07-23:
Hmm I don’t know why I didn’t think of that. I have to do that for esp32-cam. Init fails every time if you don’t deinit and deinit fails if you don’t init lol.

I will try that but at least two times it happened at initial power on. One from computer, other from battery pack. It’s possible it solves those too. I probably powered on 6 times and uploaded main and soft rebooted 50+ yesterday had no failures. Was inside yesterday vs outside with all previous failures. Air temp 80-89F, decent amount of humidity.

PsuFan, 2025-08-13:
```
>>> cam=Camera()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-261, 'ESP_ERR_NOT_FOUND')

cnadler86, 2025-08-15:
The latest release has a bunch of PRs in the low level driver that seems to bring improvements especially for jpeg capture. You might want to give it a try. 

Beside this, it is not possible to reproduce your bug. So, if you don't mind, I will close it.

PsuFan, 2025-08-15:
Okay I’ll give it a try thanks.

closed 2025-08-15 by PsuFan